STMicroelectronics/STM32CubeH7 #294: JPEG utils not setting ConvertedDataCount variable in YCbCr to RGB functions

luismeruje, 2024-06-25:
## System info
Board: Nucleo-STM32H743
IDE: STM32CubeIDE

## Bug
In the jpeg_utils.c file, which can be found under `Utilities/JPEG`, the `ConvertedDataCount` variable is not being set in the following functions:

```
static uint32_t JPEG_MCU_YCbCr420_ARGB_ConvertBlocks(uint8_t *pInBuffer, 
                                      uint8_t *pOutBuffer, 
                                      uint32_t BlockIndex,
                                      uint32_t DataCount,
                                      uint32_t *ConvertedDataCount);

static uint32_t JPEG_MCU_YCbCr422_ARGB_ConvertBlocks(uint8_t *pInBuffer, 
                                      uint8_t *pOutBuffer, 
                                      uint32_t BlockIndex,
                                      uint32_t DataCount,
                                      uint32_t *ConvertedDataCount);

static uint32_t JPEG_MCU_YCbCr444_ARGB_ConvertBlocks(uint8_t *pInBuffer, 
                                      uint8_t *pOutBuffer, 
                                      uint32_t BlockIndex,
                                      uint32_t DataCount,
                                      uint32_t *ConvertedDataCount);

static uint32_t JPEG_MCU_Gray_ARGB_ConvertBlocks(uint8_t *pInBuffer, 
                                      uint8_t *pOutBuffer, 
                                      uint32_t BlockIndex,
                                      uint32_t DataCount,
                                      uint32_t *ConvertedDataCount);

static uint32_t JPEG_MCU_YCCK_ARGB_ConvertBlocks(uint8_t *pInBuffer, 
                                      uint8_t *pOutBuffer, 
                                      uint32_t BlockIndex,
                                      uint32_t DataCount,
                                      uint32_t *ConvertedDataCount);
```

## Reprocuding the error
1. A simple code inspection is sufficient to see that the variable is not being set at any point in these functions.

## Solution
Just set the value accordingly. I believe the following lines of code could work for the case of JPEG_ARGB8888, but I'm not entirely sure how to handle the remaining RGB data formats:
```
#if (JPEG_RGB_FORMAT == JPEG_ARGB8888)
  ConvertedDataCount = numberMCU * 8 * 8 * 4; //I.e., numberMCUS * 8 * 8 block size * 4 color components
#elif(JPEG_RGB_FORMAT == JPEG_RGB888)
  ConvertedDataCount = numberMCU * 8 * 8 * 3 (?); //I.e., numberMCUS * 8 * 8 block size * 3 color components
#elif (JPEG_RGB_FORMAT == JPEG_RGB565)
  ConvertedDataCount = ?
#endif
     
```

KRASTM, 2024-06-27:
ST Internal Reference: 185221

KRASTM, 2024-07-24:
Fixed in commit https://github.com/STMicroelectronics/STM32CubeH7/commit/f164ae776466c8a099f45faefa5c58049b9d90a3

closed 2024-07-24 by KRASTM