STMicroelectronics/STM32CubeH7 #296: FatFS Outdated

squintz, 2024-08-13:
The release notes for chans FatFS list all the enhancements and fixes. From my understanding updating is not as simple as grabbing the new version from chan. We surely would appreciate if an official ST version was released.

Pleae see the release notes for a full list of fixes and enhancements.
[http://elm-chan.org/fsw/ff/updates.html](http://elm-chan.org/fsw/ff/updates.html )

This repo is current using R0.12c and R0.15 is the current version.

TOUNSTM, 2024-08-15:
Hello @squintz,

You can find the official r0.15 version at this link: [STM32 FatFS r0.15](https://github.com/STMicroelectronics/stm32_mw_fatfs/tree/master). However, please note that we have no plans to update the FatFS stack for STM32CubeH7 due to a compatibility break between version r0.14 and earlier versions.
Please allow me to close this issue now. Thank you again for your contribution.

With regards,

TOUNSTM, 2024-08-15:
Hello @squintz,

You can find the official r0.15 version at this link: [STM32 FatFS r0.15](https://github.com/STMicroelectronics/stm32_mw_fatfs/tree/master). However, please note that we have no plans to update the FatFS stack for STM32CubeH7 due to a compatibility break between version r0.14 and earlier versions.
Please allow me to close this issue now. Thank you for your contribution.

With regards,

closed 2024-08-15 by TOUNSTM