STMicroelectronics/STM32CubeH7 #298: H743IIT6 USB DMA and Dcache working together produce anomalies

andy817501, 2024-09-20:
**Describe the set-up**
 * stm32h743iit6
 * usbx
 * use USB_OTG_HS as host , and USB_OTG_FS as device
 *  USB_OTG_HS  dma_enable = ENABLE  and enable the dcache
 * vscode gcc13.2.0

**Additional context**
when i used the  usb_host and opened the dma and dcache, the usb cannot work I have divided the memory area used by USB DMA into RAM-D2, and when I turn off DCAHCE, USB host and dma can be used normally,or I can use SCB->CACR |= SCB_CACR_FORCEWT_Msk And use __DSB() in the HAL_SCD_ShenfyURBChange_Callback function; It can also be used, but I'm not sure if it will affect performance. Is there any way not to turn off dcache and not to use SCB->CACR |= SCB_CACR_FORCEWT_Msk for dcache.

andy817501, 2024-09-20:
I have configured R2 MPU_ACCESS_NOT_BUFFERABLE ， MPU_ACCESS_NOT_CACHEABLE ， MPU_ACCESS_NOT_SHAREABLE and 
MPU_TEX_LEVEL1

andy817501, 2024-09-20:
是否对于PCD_HandleTypeDef hpcd 句柄也需要分配到RAM_R2

andy817501, 2024-09-20:
But I use USB_OTG_FS dma enable=ENABLE and turn on dcache to make it work as a cdc device, It can work normally,I don't need to turn off dcache or use SCB->CACR |= SCB_CACR_FORCEWT_Msk, but usb_host can not work , i find when i  open the dcache and dma  in usb host the SETUP transaction sent to device is malformed

ALABSTM, 2024-09-23:
Hi @andy817501,

From you setup description, I see `ThreadX` and `USBX`. Are you trying to port these libraries to the STM32CubeH7 firmware? If yes, you can have a look to the [x-cube-azrtos-h7](https://github.com/STMicroelectronics/x-cube-azrtos-h7) repository. You may find the answer to your question by browsing into the different applications available.

Please keep me informed.

With regards,

andy817501, 2024-09-24:
@ALABSTM I discovered after porting that USB_cdc_host DMA and Dcache are working together to produce anomalies,but the usb_cdc_device is no such issue

ALABSTM, 2024-10-07:
Hi @andy817501,

Have you had a look at the [x-cube-azrtos-h7](https://github.com/STMicroelectronics/x-cube-azrtos-h7) repository?

With regards,

andy817501, 2024-10-08:
yes，i had a look at the [x-cube-azrtos-h7](https://github.com/STMicroelectronics/x-cube-azrtos-h7) repository , and make sure The DTCM (0x20000000) memory region do not be used ,and make sure to configure the USB pool memory region with attribute "Non-Cacheable" to ensure coherency between CPU and USB DMA

ALABSTM, 2024-10-08:
Hi @andy817501,

I forwarded your question to our development teams. Let's wait for their feedback.

With regards,

andy817501, 2024-10-10:
我参考过这个issues，但是不能解决，期待你的回复

andy817501, 2024-10-10:
I have referred to this issue, but it cannot be resolved. Looking forward to your reply

andy817501, 2024-10-10:
I had look this issue, but it only Mentioning the closure of dache，and use not cacheable Memory ，for the closure of dache i think it is unacceptable , for  not cacheable Memory ,i had use it,But there will still be problems

andy817501, 2024-10-10:
I had look this issue, but it only Mentioning the closure of dache，and use not cacheable Memory ，for the closure of dache i think it is unacceptable , for not cacheable Memory ,i had use it,But there will still be problems, We look forward to your reply

andy817501, 2024-10-23:
HI! @ALABSTM Tell me if you need any help or further information.

ALABSTM, 2024-10-31:
Hi @andy817501,

You mentioned configuring a memory region to be not cacheable. Did you ensure that your USB buffers are also in this non-cacheable region? Should I understand you did, when reading the sentence _"make sure to configure the USB pool memory region with attribute "Non-Cacheable""_?

If needed, you can refer to the provided USB examples showing how to configure the MPU, as in the code snippet below:

https://github.com/STMicroelectronics/STM32CubeH7/blob/d0e2e3886ee013b56082e57f48ae9bc6c34c849a/Projects/STM32H743I-EVAL/Applications/USB_Host/CDC_Standalone/Src/main.c#L288-L343

Please let me know whether it helps.

With regards,

andy817501, 2024-11-01:
yes, I  Configure the MPU attributes as Normal Non Cacheable for  RAM_D2,and use RAM_D2 as the usb_cdc_device's dma buffers is permissible,but I can use the usb_cdc_host

andy817501, 2024-11-01:
yes, I Configure the MPU attributes as Normal Non Cacheable for RAM_D2,and use RAM_D2 as the usb_cdc_device's dma buffers is permissible,but I can not use the usb_cdc_host with dma

closed 2024-12-26 by andy817501

ALABSTM, 2025-01-06:
Hi @andy817501,

I see you closed this issue as completed. Did you find a solution to the problem? If yes, would you mind sharing a summary please?

With regards,