STMicroelectronics/STM32CubeH7 #301: D1CorePrescTable[16] causes the RCC module to work abnormally.

hclcq, 2024-12-05:
Environment:
STM32H743VGT6 with Keil V5 HAL project generated by CUBEMX, the lab version is STM32Cube FW_H7 V1.12.0

Bug phenomenon:
The baud rate of the USART peripheral is half as low as expected.

Cause of bug(Maybe?):
[STM32CubeH7](https://github.com/STMicroelectronics/STM32CubeH7/tree/master)/[Projects](https://github.com/STMicroelectronics/STM32CubeH7/tree/master/Projects)/[STM32H743I-EVAL](https://github.com/STMicroelectronics/STM32CubeH7/tree/master/Projects/STM32H743I-EVAL)/[Templates_LL](https://github.com/STMicroelectronics/STM32CubeH7/tree/master/Projects/STM32H743I-EVAL/Templates_LL)/[Src](https://github.com/STMicroelectronics/STM32CubeH7/tree/master/Projects/STM32H743I-EVAL/Templates_LL/Src)
/system_stm32h7xx.c
The old version array D1CorePrescTable[16] = {0, 0, 0, 0, 1, 2, 3, 4, 1, 2, 3, 4, 6, 7, 8, 9} replaced by D1CorePrescTable[16] = {0, 0, 0, 0, 0, 0, 0, 0, 1, 2, 3, 4, 6, 7, 8, 9}; 

uint32_t HAL_RCC_GetPCLK2Freq(void)
{
  /* Get HCLK source and Compute PCLK1 frequency ---------------------------*/
#if defined(RCC_D2CFGR_D2PPRE2)
  return (HAL_RCC_GetHCLKFreq() >> ((D1CorePrescTable[(RCC->D2CFGR & RCC_D2CFGR_D2PPRE2)>> RCC_D2CFGR_D2PPRE2_Pos]) & 0x1FU));
#else
  return (HAL_RCC_GetHCLKFreq() >> ((D1CorePrescTable[(RCC->CDCFGR2 & RCC_CDCFGR2_CDPPRE2)>> RCC_CDCFGR2_CDPPRE2_Pos]) & 0x1FU));
#endif
}

The function above uses the D2PPRE2 bit of the D2CFGR register to calculate the PCLK2 frequency by looking up the D1CorePrescTable.However, since the D2PPRE2 only has 3 bits, the new D1CorePrescTable cannot return the correct right shift count.
PCLK2 results in a frequency that is doubled compared to its real value.
Then the baudrate calculated by the wrong input clock frequency.

Bug fix(effective for us):
Revert D1CorePrescTable[] to last version.

ALABSTM, 2024-12-06:
Hi @hclcq,

Thank you for having reported. We are already aware of this issue. A fix will be made available in the frame of release **v1.12.1**, soon enough hopefully.

With regards,

KRASTM, 2024-12-11:
Hello @hclcq,

The fix is already available within the frame of release [v1.12.1](https://github.com/STMicroelectronics/STM32CubeH7/commit/f5c0b7a2b1f6eb26fde150f72edb2d7deb647066)

Thank you again,

With regards

closed 2024-12-11 by KRASTM

ALABSTM, 2026-02-10:
ST Internal Reference: 182575

hiwayWang3rd, 2026-02-17:
> Hi @hiwayWang3rd<https://github.com/hiwayWang3rd>,
>
> Please note that in release 1.12.1, the D1CorePrescTable[] table has been reverted to its original content (i.e., {0, 0, 0, 0, 1, 2, 3, 4, 1, 2, 3, 4, 6, 7, 8, 9}) as you can see here<https://github.com/STMicroelectronics/cmsis-device-h7/blob/e8d40ae6e2fa06afe5b46d24756f141b363342a7/Source/Templates/system_stm32h7xx.c#L155>, which has solved the issue.
>
> With regards,

Thanks for the information, have a good day.