STMicroelectronics/STM32CubeH7 #302: stm32h7xx_hal_eth.c bug report

Gorden1970, 2024-12-19:
STM32H7 HAL Library version 1.12.0 and 1.12.1 stm32h7xx_hal_eth.c line number 2503 content is as follows

 ((uint32_t)((pFilterConfig->BroadcastFilter == DISABLE) ? 1U : 0U) << 5) |

STM32H7 HAL Library version 1.11.0 and 1.11.1 stm32h7xx_hal_eth.c line number 2503 content is as follows

 ((uint32_t)((pFilterConfig->BroadcastFilter == ENABLE) ? 1U : 0U) << 5) |

STM32H7 HAL Library version 1.12.0 and 1.12.1 will cause ETH to be unusable, I check file RM0399 description

Bit 5 DBF: Disable Broadcast Packets
When this bit is set, the AFM module blocks all incoming broadcast packets. In addition, it
overrides all other filter settings.
When this bit is reset, the AFM module passes all received broadcast packets.

mikhail11111, 2024-12-20:
In version 1.12.1 I have it as ENABLE, but Ethernet still doesn't work in the program that worked with 1.11.1.

ASEHSTM, 2024-12-24:
Hello @Gorden1970,

Thank you for your report. However, referring to the reference manual, when bit 5 is set, broadcast packets are disabled, which means that the BroadcastFilter is enabled. Therefore, the correct code is from version 1.12.0 and 1.12.1:

  `((uint32_t)((pFilterConfig->BroadcastFilter == ENABLE) ? 1U : 0U) << 5)`

This ensures that when the BroadcastFilter is enabled, bit 5 is set to 1 to block broadcast packets. Conversely, when the BroadcastFilter is disabled, bit 5 is set to 0 to allow all broadcast packets to pass.

With Regards,

Gorden1970, 2024-12-25:
I agree with this and think what you said is right, but the drivers currently provided by Keil MDK are as follows

Driver_ETH_MAC.h
  #define ARM_ETH_MAC_ADDRESS_BROADCAST (1UL << 7) ///< Accept frames with Broadcast address

ETH_MAC_STM32.c
  if ((arg & ARM_ETH_MAC_ADDRESS_BROADCAST) != 0U) {            // --- Broadcast Frame address
    eth_mac0_rw_info.mac_filter.BroadcastFilter = ENABLE;
  } else {
    eth_mac0_rw_info.mac_filter.BroadcastFilter = DISABLE;
  }

This will cause existing projects to not function properly on V1.12.1. I can only modify to DISABLE during transition.

And thank you for your reply

ASEHSTM, 2025-01-13:
Hello @Gorden1970,

This issue is related to the driver provided by Keil MDK. On our side, the correct code is from versions 1.12.0 and 1.12.1. 

You can report the issue on this forum: https://github.com/Open-CMSIS-Pack/CMSIS-Driver_STM32/blob/main/Drivers/ETH_MAC_STM32.c 

Thank you.
With Regards,

ASEHSTM, 2025-01-17:
Hello @Gorden1970,

Please allow me to close this ticket.

With Regards,