STMicroelectronics/STM32CubeH7 #303: STM32H750XBH6 sends error data when using UART4 of the 120M clock in Latest version(STM32Cube FW H7 V1.12.1)

augetyz, 2024-12-23:
STM32H750XBH6 sends error data when using UART4 of the 120M clock in latest version(STM32Cube FW H7 V1.12.1).But in STM32Cube FW H7 V1.11.1,the same code has no errors.Cubemx version is latest version v6.13.0.When I make UART4 clock descend to 75M，the error is ok.In STM32Cube FW H7 V1.11.1,the UART4 clock do work well in 120M clock.

Gorden1970, 2024-12-23:
I'm using H7 V1.12.1/V1.11.1 on STM32h755/STM32h743, 120M clock source, baud rate 921600, it works fine

KRASTM, 2024-12-27:
Hello @augetyz,

Thank you for the report.

Could you please, provide us with more details about the error (maybe a screenshot in debug mode), and your ioc file. In the meantime, I will try to do some tests.

With regards,

KRASTM, 2025-01-13:
Hello @augetyz,

Since there is no activity, please allow me to close this thread. You may reopen it at any time if you have any details to share with us in order to help you to solve the issue.

Thank you for your comprehension.

With regards,

closed 2025-01-13 by KRASTM