STMicroelectronics/STM32CubeH7 #304: STM32H7 USB, SD Card and ADC problem

vahidajalluian, 2025-01-08:
Since my forum account has been banned, I'm reporting bugs here instead.

**Describe the set-up**
I am using a custom board with the following specifications:

MCU: STM32H743BIT6
STM32CubeMX version: 6.13
H7 firmware version: V1.12.0

**Describe the bug**
 When one create project from scratch with stm32cubemx these bugs will appear.

1- USB Device Descriptor Issue with Optimizations
 I had also this problem with f4 series, when enabling optimization (o1 and above), USB device descriptor cannot be estimated, and therefore no device is presented Device manager of windows. 
Solution: Modify the Get_SerialNum function in usbd_desc.c as follows:

```
static void Get_SerialNum(void)
{
  uint32_t deviceserial0;
  uint32_t deviceserial1;
  uint32_t deviceserial2;

  deviceserial0 = *(uint32_t *) DEVICE_ID1;
  deviceserial1 = *(uint32_t *) DEVICE_ID2;
  deviceserial2 = *(uint32_t *) DEVICE_ID3;

  deviceserial0 += deviceserial2;

//  if (deviceserial0 != 0)
//  {
//    IntToUnicode(deviceserial0, &USBD_StringSerial[2], 8);
//    IntToUnicode(deviceserial1, &USBD_StringSerial[18], 4);
//  }

  USBD_StringSerial[2] = 'D';
  USBD_StringSerial[4] = 'E';
  USBD_StringSerial[6] = 'V';
  USBD_StringSerial[8] = 'M';
  USBD_StringSerial[10] = 'E';
}
```

2-USB HS Pin Configuration Issue

Enabling USB HS in device-only mode forces PB13 to take on an alternate function, but the specific function is unclear. Despite configuring PB13 as an input, the pin behavior was erratic, and how hard I have tried, I couldn't understand what is the problem, so I switched to USB FS peripheral.

3- Enabling USB does not automatically set the USB clock to 48 MHz, causing the USB to be unrecognizable by endpoint systems.

4- Enabling SDMMC and Fatfs, I could not make it work and emulate the SDCARD. However in F4 series program I had earlier, I saw that it was forcibly enabled NVIC, so I had enabled the NVIC and the problem solved.

5- (IOC file included) In ADC, I had set the resolution to be 12 bit for both ADC1 and ADC3, but only ADC1 is affected and ADC3 is generated without setting(and so the code presume it as 16 bit) although I have checked and it has been set to be 12 bit in IOC file.

[ADC_H7.zip](https://github.com/user-attachments/files/18343241/ADC_H7.zip)

vahidajalluian, 2025-01-14:
@ALABSTM I have created two new issues for item 4 and 5.
And for those who don't know why my account has been disabled, It has been disabled, purely for being in Iran :( .

vahidajalluian, 2025-02-01:
@ALABSTM Sorry, I had misleadingly said PB13, but it is PB12. Sorry for my mistake.

vahidajalluian, 2025-04-02:
@ALABSTM any updates on this issue?
Could you reproduce Usb problems?

ALABSTM, 2025-07-31:
Hello @vahidajalluian,

I hope you are doing well. Below the feedback from our development team regarding the USB-related points you reported.

1. We tested the original `Get_SerialNum()` with optimization level O1 and above using both H7 and F4 MCU and encountered **no issue**. The issue you faced is likely to be due to something else. 
 
2. Actually, the **internal PHY** of the STM32H743 can generate **only full speed** signals. To reach high speed, there is a need for an **external PHY** chip. The interface with such external PHY chip is called **ULPI**. This interface requires **12** GPIO, among which **PB12** and **PB13**. This is why both pins are configured as alternate functions **when the HS is selected**. For more details, you can refer to:
    * table 9 of [ DS12110](https://www.st.com/resource/en/datasheet/stm32h743bi.pdf) ( Rev. 10).
    * section "57.4.6 High-speed OTG PHY connected to OTG_HS" of [RM0433](https://www.st.com/resource/en/reference_manual/rm0433-stm32h742-stm32h743753-and-stm32h750-value-line-advanced-armbased-32bit-mcus-stmicroelectronics.pdf) (Rev. 8)
 
3. Depending on the peripherals enabled and your configuration, it may happen that the STM32CubeMX cannot resolve some aspects of the clock tree. In such case, you have to do it manually.

I hope this clarifies the different points. Please allow me to close this issue.

With regards,

closed 2025-07-31 by ALABSTM

vahidajalluian, 2025-07-31:
Would you please also test it with o3?
I had to set it to o3 since I was using LVGL