STMicroelectronics/STM32CubeH7 #305: STM32H7 CUBEMX SD Card FATFS NVIC problem

vahidajalluian, 2025-01-14:
Following the request in this [issue](https://github.com/STMicroelectronics/STM32CubeH7/issues/304), I have created this one.

Describe the set-up
I am using a custom board with the following specifications:

MCU: STM32H743BIT6
STM32CubeMX version: 6.13
H7 firmware version: V1.12.0

Describe the bug
When one creates a project from scratch with stm32cubemx this issue will happen.

Enabling SDMMC and Fatfs, I could not make it work and emulate the SDCARD. However, in the F4 series program I had earlier, I saw that it was forcibly enabled NVIC, so I enabled the NVIC and the problem was solved.

KRASTM, 2025-01-16:
Hello @vahidajalluian,

Thank you for the report.

Could you please, share your .ioc file, and more details about the error?

With regards,

vahidajalluian, 2025-01-16:
I will share a checked ioc to reproduce the issue by sunday. I am not at work these two days.

vahidajalluian, 2025-02-03:
> Hello @vahidajalluian <https://github.com/vahidajalluian>,
>
> Any news please?
>
> With regards

Sorry, I was away and i had forgotten this one totally, reaching home, I
will send an ioc file.

vahidajalluian, 2025-02-05:
@ALABSTM @KRASTM 
Here is the example code with IOC that doesn't force activation of NVIC. Sorry for being (too much) late
[sdTest.zip](https://github.com/user-attachments/files/18667161/sdTest.zip)

KRASTM, 2025-02-07:
Hello @vahidajalluian,

Regarding your issue, I think there is a misunderstanding. The SDMMC driver supports 3 modes: polling, interrupt and DMA like almost all IP, for the NVIC and as mentioned in the brief of the driver file, it enabled when interrupt process used.

> (#) Initialize the SDMMC low level resources by implementing the HAL_SD_MspInit() API:
        (##) Enable the SDMMC interface clock using __HAL_RCC_SDMMC_CLK_ENABLE();
        (##) SDMMC pins configuration for SD card
            (+++) Enable the clock for the SDMMC GPIOs using the functions __HAL_RCC_GPIOx_CLK_ENABLE();
            (+++) Configure these SDMMC pins as alternate function pull-up using HAL_GPIO_Init()
                  and according to your pin assignment.
        **(##) NVIC configuration** if you need to use interrupt process (HAL_SD_ReadBlocks_IT()
             and HAL_SD_WriteBlocks_IT() APIs).
            (+++) Configure the SDMMC interrupt priorities using function HAL_NVIC_SetPriority();
            (+++) Enable the NVIC SDMMC IRQs using function HAL_NVIC_EnableIRQ()
            (+++) SDMMC interrupts are managed using the macros __HAL_SD_ENABLE_IT()
                  and __HAL_SD_DISABLE_IT() inside the communication process.
            (+++) SDMMC interrupts pending bits are managed using the macros __HAL_SD_GET_IT()
                  and __HAL_SD_CLEAR_IT()
        (##) No general propose DMA Configuration is needed, an Internal DMA for SDMMC Peripheral are used.

I tried the same config with F4 Serie, and I didn't find that `it was forcibly enabled NVIC`, I think it's a choice for the user to enable or not NVIC and may depend on the application. 

If you could share your API sequence used in your project for further analysis, otherwise you can check examples and/or application with DK and Eval boards under projects for further details.

With regards,

vahidajalluian, 2025-02-07:
@KRASTM Thank you for your follow-up,
Using SDIO alone doesn't forcibly require enabling the NVIC of SDIO, but enabling fatfs will automatically force SDIO to activate NVIC generation.

vahidajalluian, 2025-02-08:
@KRASTM Thank you for your follow up
My mistake about activating NVIC forcibly, but if I don't activate it, It wont work with Fatfs, and reason is unknown for me.
[image]

KRASTM, 2025-02-10:
Hello @vahidajalluian,

Since your application includes other IP such as USB and ADC beside SDMMC and FATFS as described in #304, also you do not activate NVIC for SDMMC (in other word you work with polling mode), maybe another interrupt is the source of your problem and to get it to work you need to enable NVIC.

I suggest you only test SDMMC and FATFS to be sure, otherwise if you could share with us your sequence and perhaps some details in debug mode.

With regards

vahidajalluian, 2025-02-13:
Dear @KRASTM with the project I have sent here using STM32H7 CUBEMX on which SD Card and FATFS are enabled, I couldnt make it work even it doesnt emulate it. But after enabling mentioned intereupt, sd card could be emulated.

KRASTM, 2025-02-17:
Hello @vahidajalluian,

Just one last request, could you please send me the same example `sdTest.zip` with NVIC enabled. I think I have found the cause, just to compare and be sure.

With regards,

vahidajalluian, 2025-02-17:
Ok, I will send it tomorrow, I am nof at the site today.

KRASTM, 2025-02-20:
Hello @vahidajalluian,

Regarding the issue, the fix will be included in the future release v6.15.0 of Cube MX or later. Please allow me to close this thread.
Thank you for your comprehension.

With regards,

closed 2025-02-21 by KRASTM