STMicroelectronics/STM32CubeH7 #306: STM32H7 ADC problem

vahidajalluian, 2025-01-14:
Following the request in this [issue](https://github.com/STMicroelectronics/STM32CubeH7/issues/304), I have created this one.

Describe the set-up
I am using a custom board with the following specifications:

MCU: STM32H743BIT6
STM32CubeMX version: 6.13
H7 firmware version: V1.12.0

(IOC file included) In ADC, I had set the resolution to be 12 bit for both ADC1 and ADC3, but only ADC1 is affected and ADC3 is generated without setting(and so the code presume it as 16 bit) although I have checked and it has been set to be 12 bit in IOC file.
[ADC_H7.zip](https://github.com/user-attachments/files/18405810/ADC_H7.zip)

RJMSTM, 2025-01-17:
Hello @vahidajalluian,

I am taking over your project and i try to understand why STM32CubeMX does not generate the resolution for ADC3. I repeat the process and attempt to configure ADC3 alone with different resolutions (8, 12, 16, etc.), but STM32CubeMX is unable to generate the resolution for ADC3 instance on STM32H743.

Unfortunately, we don't treat aspect related to STM32CubeMX tool in our GitHub repositories. They are rather treated at the [STM32CubeMX](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx) dedicated page of the ST Community.

Since this issue is not directly related to the STM32Cube firmware but rather to our ecosystem, please allow me then to close this thread. Thank you for your comprehension.

With regards,

vahidajalluian, 2025-01-17:
My account has been suspended in the forum. If my account hasnt been suspended, I would have opened an issue on the forum, so if you can please inform them to solve this issue. and since you are saying that this is not releated to this repository but to the ecosystem, it is ok to close this issue.

RJMSTM, 2025-02-05:
Hello @vahidajalluian,

Please don't worry. I have reported your issue to the STM32CubeMX team, and I am hopeful that it will be addressed in an upcoming release.

Regards,

RJMSTM, 2025-02-05:
ST Internal Reference: 202321

closed 2025-02-05 by RJMSTM