STMicroelectronics/STM32CubeH7 #311: Issue with building example for for H723ZG - Lwip - TCP and UDP examples

surendersampath, 2025-03-10:
**Caution**
I'm unable to build this project using the Keil compiler 5.02 or 6.

**Describe the set-up**
 * Using Nucleo Board
 * IDE or at least the compiler and its version

**Additional context**

[image]

**Screenshots**
If applicable, add screenshots to help explain your problem.

ASEHSTM, 2025-03-14:
Hello @surendersampath ,

Thank you for your contribution. On our side, there are no errors in these projects. Have you tried other H7 projects? Do you encounter the same issue with all H7 projects?

It looks like there might be an issue with file paths on your end. Please verify that the paths are correctly set and ensure the specified directories are accessible, as shown in the image below.

[image]

and please let me know the version of Keil you are using.

With Regards,

ASEHSTM, 2025-03-24:
Hello @surendersampath,

Any update regarding this thread?

With Regards,

surendersampath, 2025-03-27:
@ASEHSTM I reverted to to using CUBE IDE. Didn't get a chance to check the path on Keil. Will try when I get a chance.

ASEHSTM, 2025-04-02:
Hello @surendersampath,

okay, please allow me to close this thread. You may reopen it at any time if you have any details to share with us.

With Regards,

closed 2025-04-02 by ASEHSTM