STMicroelectronics/STM32CubeH7 #313: Wrong selections in stm32h743 hrtim's dead time settings

wangyz1997, 2025-03-20:
**Describe the bug**

STM32CubeMX version: 6.14.0
STM32CubeMX DB version: DB.6.0.140
STM32CubeH7 firmware version: STM32Cube FW_H7 V1.12.1
Chip: STM32H743IIT6

When configuring htrim's dead time, "Prescaler" has selections like "fDTG = fHRTIM * 8" and "fDTG = fHRTIM * 4", which is not present in the version of the peripheral on STM32H743, causing a compilation error.

**How To Reproduce**

1. Enable htrim on STM32H743
2. Activate channels of hrtim
3. Go to the tab of enabled channel
4. Under "Timing Unit", change "Dead Time Insertion" to "Deadtime is inserted between output 1 and output 2"
5. Now "Dead Time" group should appear. Under "Dead Time", the default selection of "Prescaler" is "fDTG = fHRTIM * 8", left it as it is
6. Generate project and build
7. The following error occurs

```
arm-none-eabi-gcc "../Core/Src/system_stm32h7xx.c" -mcpu=cortex-m7 -std=gnu11 -g3 -DDEBUG -DUSE_PWR_LDO_SUPPLY -DUSE_HAL_DRIVER -DSTM32H743xx -c -I../Core/Inc -I../Drivers/STM32H7xx_HAL_Driver/Inc -I../Drivers/STM32H7xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32H7xx/Include -I../Drivers/CMSIS/Include -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -fcyclomatic-complexity -MMD -MP -MF"Core/Src/system_stm32h7xx.d" -MT"Core/Src/system_stm32h7xx.o" --specs=nano.specs -mfpu=fpv5-d16 -mfloat-abi=hard -mthumb -o "Core/Src/system_stm32h7xx.o"
../Core/Src/main.c: In function 'MX_HRTIM_Init':
../Core/Src/main.c:271:28: error: 'HRTIM_TIMDEADTIME_PRESCALERRATIO_MUL8' undeclared (first use in this function); did you mean 'HRTIM_TIMDEADTIME_PRESCALERRATIO_DIV8'?
  271 |   pDeadTimeCfg.Prescaler = HRTIM_TIMDEADTIME_PRESCALERRATIO_MUL8;
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                            HRTIM_TIMDEADTIME_PRESCALERRATIO_DIV8
../Core/Src/main.c:271:28: note: each undeclared identifier is reported only once for each function it appears in
make: *** [Core/Src/subdir.mk:34: Core/Src/main.o] Error 1
make: *** Waiting for unfinished jobs....
```

**Screenshots**

[image]

KRASTM, 2025-03-24:
ST Internal Reference: 205967

KRASTM, 2025-04-14:
Hello @wangyz1997,

Regarding the issue, the fix will be included in the future release v6.15.0 of Cube MX or later. Please allow me to close this thread.
Thank you for your comprehension.

With regards,

closed 2025-04-14 by KRASTM