STMicroelectronics/STM32CubeH7 #314: STM32CubeH7 1.12.1 SAI4 configuration is broken

BlockWorker, 2025-04-09:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus) 

**Describe the set-up**
Custom board using an STM32H725RGV6 controller with STM32CubeIDE v1.18.0.

**Describe the bug**
In the .ioc file editor, the SAI4 peripheral configuration for STM32H725RGV6 cannot be edited (editor doesn't load the configuration options). Attempting to generate code with the SAI4 peripheral enabled also shows an error about the configuration being incorrect.

**How To Reproduce**
Create a new STM32 project in STM32CubeIDE v1.18.0, targetting STM32H725RGV6, with H7_FW v1.12.1.
In the .ioc file editor, enable the SAI4 peripheral (e.g., SAI B to "Master" mode). The configuration options window doesn't load.

**Screenshots**

[image]
[image]

KRASTM, 2025-04-11:
ST Internal Reference: 207476

KRASTM, 2025-05-02:
Hello @BlockWorker,

Regarding the issue, the fix will be included in the future release v6.15.0 of Cube MX or later. Please allow me to close this thread.
Thank you for your comprehension.

With regards,

closed 2025-05-02 by KRASTM