STMicroelectronics/STM32CubeH7 #321: STM32H74xI/G Fails to Initialize System Clock Config with Generated Code by STM32CubeMX

littlebutt, 2025-07-12:
**Describe the set-up**
 * STM32H743XIHx
 * arm-none-eabi-gcc  (Arm GNU Toolchain 14.2.Rel1 (Build arm-14.52)) 14.2.1 20241119
 * openocd 0.12.0 (2024-09-16)

**Describe the bug**
When I used the default configuration generated by STM32CubeMX(with the highest clock frequency - 480Mhz), st-link loss its connection. While debugging with gdb, openocd complained that `Error: jtag status contains invalid mode value - communication failure`when it ran into the code as follows.
```
__HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE1); // Or PWR_REGULATOR_VOLTAGE_SCALE2, but PWR_REGULATOR_VOLTAGE_SCALE3 is OK
```

It seems that HAL cannot set __VCORE__ bit in the reg and make the st-link fail to connect to the chip. As the result I have no way but to downgrade the clock frequency to 200MHz.

**How To Reproduce**
1. New a project with STM32CubeMX and Set the clock frequency higher than 200MHz.
2. Compile and download the generated code.
3. After download it once successfully, you cannot download it twice (st-link loses the connection).

**Additional context**
gdb stack
```
(gdb) c
Continuing.

Breakpoint 1, SystemClock_Config () at Core/Src/main.c:122
122       HAL_PWREx_ConfigSupply(PWR_LDO_SUPPLY);
(gdb) n
halted: PC: 0x08000426
halted: PC: 0x08002110
126       __HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE1);
(gdb) s
halted: PC: 0x0800042c
halted: PC: 0x0800042e
halted: PC: 0x08000430
halted: PC: 0x08000432
halted: PC: 0x08000436
halted: PC: 0x08000438
halted: PC: 0x0800043a
halted: PC: 0x0800043e
halted: PC: 0x08000440
halted: PC: 0x08000442
halted: PC: 0x08000444
halted: PC: 0x08000448
jtag status contains invalid mode value - communication failure
Previous state query failed, trying to reconnect
jtag status contains invalid mode value - communication failure
Previous state query failed, trying to reconnect
jtag status contains invalid mode value - communication failure
Previous state query failed, trying to reconnect
jtag status contains invalid mode value - communication failure
Previous state query failed, trying to reconnect
...
```

KRASTM, 2025-07-14:
Hello @littlebutt,

Thank you for the report.

Could you please tell us which version of CubeMx you are using? If possible, could you also share your .IOC file in order to check it and reproduce the issue?

With regards,

KRASTM, 2025-08-12:
Hello @littlebutt,

Please, allow me to close this issue. Thank you for your understanding.

With regards,